newaetech/chipshouter-picoemp #1: Kit / Easy-Assemble Build Availability

colinoflynn, 2021-12-21:
One goal is to eventually offer a "partial kit", which will be in $50 to $100 range most likely. This kit would require you to solder the R-Pi Pico & some switches onto the board, but the SMD part would be **done**. This board would also be hi-pot + functionally tested,

*NOTE: A "full assembly" kit is probably never going to be offered, as the cost of kitting the SMD parts would be more than the cost of the SMD run. SMD kits have a higher failure rate which adds considerable "average support costs" ontop. Due to the availability of very low cost PCB services like JLCPCB or OSHPARK if you want a full kit please order the PCB & parts directly.*

gmgunderground, 2021-12-22:
Well done, I will wait for it!

SoftwareEngineer, 2021-12-22:
Nice! Very interested in a "partial kit". Will it be possible to put a header on the board instead of soldering the R-Pi pico right on the board?

david-oswald, 2021-12-23:
Definitely interested in 1-2 of them.

inode-, 2021-12-23:
Waiting for it!

grymoire, 2021-12-24:
Put me down for a maybe.

matthew5025, 2021-12-25:
+1 would be interested too!

m-ober, 2021-12-26:
Count me in.

jcldf, 2021-12-26:
yeah! I'm interested

avloginov, 2021-12-26:
I'd be happy to buy either easy to assemble or just empty board.

osok, 2021-12-28:
I am interested

FlUxIuS, 2021-12-29:
Would be happy to test the easy-assemble build! 
Thanks in advance!

doegox, 2021-12-30:
Yup, I'm interested by the easy-assemble kit too :)

jonahedwards, 2022-01-03:
Definitely interested!

RobinFassina-Moschini, 2022-01-05:
Interested by the easy-assemble kit too.

KypForty, 2022-01-07:
Interested by mostly complete SMD board. Regards!

jslawek, 2022-01-13:
[image]

gohai, 2022-01-16:
Interested in partial kit

AlexanderKrampe, 2022-01-17:
Interested +1

shinmai, 2022-01-18:
+1 for sure

LovelyA72, 2022-01-18:
Sign me up!

gworona, 2022-01-20:
Interested

Gabbo28, 2022-01-21:
Interested!

max-r-b, 2022-01-25:
Interested as well! Thanks!

hexsecs, 2022-01-25:
Definitely interested!

Xenador77, 2022-02-28:
Interested as a southerner in a colloquialism

hgordon-git, 2022-03-08:
super interested. I would get one to three.

pr0z78, 2022-03-15:
Yes hyped and interested for at least 2 (shared buy for co-worker) or 3 (also 1 for team) when it  becomes available.

afflitto, 2022-04-05:
very interested!

colinoflynn, 2022-05-30:
Thanks for the patience all - we've released our first run of the kit at https://store.newae.com/chipshouter-picoemp . If you'll be at RECON in Montreal we have a special discount - free shipping if you pick up in person (use code RECON). Note the following:

* Requires 2xAA batteries for use (excluded as we weren't sure if it caused shipping issues, it looks like maybe not but we hadn't ordered them for the kit builds so we don't want to delay on that)
* We ship via DHL - we cannot ship via mail as historically the service is very unreliable & it's not worthwhile for us. This makes the shipping cost a large % of the unit cost - you can save a bit by splitting shipping, 4 fit in our smallest box.

We'll eventually see if there is easier ways to stock this - but for now in this initial run the logistics are likely going to be a bit heavy!

autoprog, 2022-11-14:
Hi, 
in Moser not have https://store.newae.com/chipshouter-picoemp
Do you ship to Brazil?,
thanks,
Carlos.

mrx23dot, 2024-02-27:
Any update on availability? What about selling on Tindie or Sparkfun, I think they fullfil orders.

eried, 2024-08-27:
:) still no mouser? :/