newaetech/chipshouter-picoemp #2: Python Interface?

colinoflynn, 2021-12-21:
Currently the simple micropython is button driven. Would be good to add a way of interacting from a computer with the board (with caveat it's "officially not recommended" to run board from computer without isolation).

lukaskuzmiak, 2021-12-23:
Hopefully not too unrelated - would you say it is a good idea run the board from the computer using a USB isolator of some sort? If you think so, maybe mention that when you end up making the interface.

At least from my point of view, I'd be happy to consider the PicoEMP and any other HW to interface with target technically as sacrificial but I'd want that strongly isolated from other (possibly a lot more expensive) equipment.

colinoflynn, 2021-12-23:
Depends how expensive your computer is ;-) I'm wondering about that side too.. using a different RP2040 board would let me add a USB isolator chip, but otherwise/for now I'd guess something like https://www.adafruit.com/product/2107 would be a good starting point. I've even seen cheaper versions (like https://www.aliexpress.com/item/4000047053305.html) that are total no-brainers! Maybe I'll add that note into the usage for now at least.

To be honest I'm running the thing directly plugged into my 1-month-old laptop, so it "should" be fairly safe (the DC-DC transformers are rated 500Vrms isolation, and the charge normally goes only to 250V DC).

But of course besides part failure, if you had a solder ball or something left over by accident it would also kill the isolation! So adding the $10 to $40 isolator cost is well worth it for anyone running from a computer.

mattytrentini, 2022-10-22:
Since the Pico W has been released - and it's pin-compatible - it would be possible to use wifi (or BLE, soon) to trigger the glitch....