newaetech/chipshouter-picoemp #5: Part Substitutions

colinoflynn, 2021-12-26:
Using this issue to track potential subs for problematic parts.

colinoflynn, 2021-12-26:
For Q3/Q4 - big important point is that it's fully "on" by 3V. There is a Schottky diode drop on the gate drive so you've got to be careful there. Check the Vgs figure for any sub - here is the "original" AO3422:

[image]

Then the PMV37ENEA (this was used in current builds):

[image]

The specific Rds isn't too important, mostly that it's "on" by that point. I've noticed some logic-level MOSFETs are only starting to turn on around 3V, so you'll see that curve shifted to the right a little more than I'd like. I haven't done much testing here so it *may* not be too critical, but if you want to avoid potential issues I'd just check the Vgs curve for anything you find in stock (along with footprints).

osok, 2022-01-02:
Where can we get the probe tips?  Is there a good resource for making them?

colinoflynn, 2022-01-02:
@osok There is a lot of experimentation needed there IMO. But you'll find some examples in https://github.com/newaetech/chipshouter-picoemp/tree/main/hardware/injection_tips for now. There are more resources "out there" (other papers) I'm figuring out best way to link, there were a few linked from the main README as well.

nilswiersma, 2022-02-05:
Is there any specific upside to having C3 being multilayered? Or as long as uF and V and height is as you mentioned in README, should be good?

colinoflynn, 2022-02-06:
By multilayered you mean the part number which has the two thin devices stacked? I wouldn't worry about that specifically. You could compare ESR/ESL specs and similar too, but for the most part I think any ceramic device is going to be good enough for picoemp usage!

PoroCYon, 2022-02-28:
I've used a SQ2364EES-T1_GE3 as replacement for Q4. Ron is a bit higher and the threshold voltage is a bit lower, but it seems to work. Sadly by now they seem to be out of stock as well.

colinoflynn, 2022-02-28:
Thanks for dropping the note still! At least good to know what sort of "range" is doable with these things (and maybe it comes back in stock later?)

gmgunderground, 2022-04-03:
Hi, no one of the C3 codes are available on Mouser, does 
[CGA9P1X7T2J474M250KC](https://www.mouser.it/ProductDetail/TDK/CGA9P1X7T2J474M250KC?qs=xLDY6iXSiQaSRDUMzY513g%3D%3D)
[CKG57NX7R2J474M500JH](https://www.mouser.it/ProductDetail/TDK/CKG57NX7R2J474M500JH?qs=NRhsANhppD%252BAmXDR5rVERg%3D%3D)
[CKG57KX7T2J474K335JH](https://www.mouser.it/ProductDetail/TDK/CKG57KX7T2J474K335JH?qs=xLDY6iXSiQaxAZWnHJ3xxg%3D%3D)
[C5750X7T2J474M250KE](https://www.mouser.it/ProductDetail/TDK/C5750X7T2J474M250KE?qs=LcTL%2F5vFEzGH3h7%252BwrjxaQ%3D%3D)
Are valid alternatives? 
These appear to be available on Mouser at the moment

colinoflynn, 2022-04-03:
They should work I think! You can double-check the size dimensions for footprint, but they look reasonable. The original part was a bit of a random pick (at the time lots of stock) so wasn't anything besides voltage/cap/case that I matched.

LennertW, 2022-04-03:
CKG57KX7T2J474K335JH worked for me!

gmgunderground, 2022-04-03:
Perfect, thank you very much

gmgunderground, 2022-05-03:
It's becoming difficult to find all the components.
I can't find RGT16BM65DTL on Mouser, is ok to substitute with IKD15N60RATMA1? Otherwise any suggestion?
Also AO3422 and alternative PMV37ENEAR are still unavailable on Moser and Digikey, what can I use as an alternative (best available on Mouser)?

gmgunderground, 2022-05-30:
Can RGPZ10BM40FHTL or IKD15N60RC2ATMA1 be a valid substitute for Q2 (RGT16BM65DTL) ? On Digikey is the only component I can't find.
What can I use? Mouser and Digikey doesn't have RGT16BM65DTL.
PCV-0-472-03L is missing too (Coil).

colinoflynn, 2022-05-31:
The coil isn't critical - in fact we ended up providing a different PN in the kit, we used Wurth 744710603. Check the `hardware/injection_tips` for some details of that.

I'd think any of these IGBTs would work ok! There wasn't anything too magic about that, mostly just the voltage rating (current too, but as the current is only a short pulse so not as critical). I think my specs for both are way overkill, so yours should be fine!

settinger, 2022-08-21:
In the Build Details page, D1 D3 D4 D5 are described as Schottky diodes, and they are depicted as such in the schematic, but SM4005PL-TP appears to be a general purpose diode. I don't see any reason why SM4005PL-TP *wouldn't* work, but figured I should point out the discrepancy. Really looking forward to building this, thank you!

colinoflynn, 2022-08-21:
Oops, thanks @settinger! Probably just when first doing the design planned on using them, but then never actually did. I'll remove that so it's not confusing people on the hunt for other parts.

colinoflynn, 2022-12-11:
FYI I tested @gmgunderground's suggested sub on Q2, IKD15N60RC2ATMA1. Seems to work, and pulse looks similar - Yellow is RGT16BM65DTL, Blue is IKD15N60RC2ATMA1. Will update the BOM as well since we're likely to use this in a few runs due to using all of our own horded stock of RGT16BM65DTL now.
[image]

gmgunderground, 2024-07-03:
R1 of 20M 1W is not available on Mouser, can the following that is 0.5W suitable too?
https://www.mouser.it/ProductDetail/Vishay-Techno/CRMA2010AF20M0FKEF?qs=1mbolxNpo8frXfdCgvcryg%3D%3D