newaetech/chipshouter-picoemp #7: Aisler: inner milling paths missing and/or incomplete

PoroCYon, 2022-01-15:
When trying to order the board from [Aisler](https://aisler.net/), a warning is shown, saying that the Gerber files don't have some required information on the inner milling paths meant for the shield. As the repository doesn't contain any EDA project files, I can't really fix it myself.

[image]

(Also, would it be possible to have a PCB that conforms to Aisler's "[simple](https://aisler.net/help/design-rules-and-specifications/design-rules)" design rules? This would reduce the cost of ordering one.)

colinoflynn, 2022-01-16:
Unfortunately it's not reasonable to create variants for all the different services out there, the existing gerbers seemed to work with at least several low-cost services tested (oshpark, pcbway, jlcpcb). Not sure where the inner milling error is - it may be this is actually milled as a slot so there is no "closed path", and the error is on their side? I'm planning on pushing the src once the cleanup is done a little more, they are Altium anyway so normally less useful.

closed 2022-01-16 by colinoflynn

hiviah, 2022-01-18:
I ordered from aisler despite those warnings and the boards arrived correctly.
[image]

invd, 2022-01-30:
My order with Aisler had different results and the inner milling paths on the PCBs **are missing** similarly to the red board variant in the [introduction video](https://www.youtube.com/watch?v=nB5arJi-tVE&t=103s). The mounting holes at the edge of the PCB as well as the two shield mounting holes near the high voltage SMA connector are done correctly.

For people in a similar situation: it is possible to cut away the notches and tabs on the Hammond enclosure and still screw it in place as a high voltage shield via the two remaining mounting holes.

On a related note: the two black screws supplied with the Hammond 1551 case are slightly too large in diameter to go through the mounting holes without biting into the PCB. From what I can tell, it is better not to stress the PCB at that point, so I used smaller screws instead. I'm not sure if this is connected to milling problems, but it's worth thinking about larger diameter mounting holes to avoid this problem in newer revisions.