newaetech/chipshouter-picoemp #8: Rpi pico dead after initial test

mehdideveloper, 2022-01-23:
Hey Colin
I built a PicoEMP today, however I was not sure how to properly test it to make sure everything is working as expected.
I uploaded the C-based firmware, and initially everything seemed to be working (LEDs working. Pushing ARM made D6 turn on, etc)
After a few times pushing the ARM and PULSE buttons, device stopped working (LEDs stayed ON. I reset the device and now LEDs don't turn on. The rpi pico's builtin LED dimly turns on and then off. I think maybe rpi is damaged. Is it possible to actually somehow damage the rpi pico?)
Now if I turn it on and wait a few minutes, the ARMING LED dimly turns on and if I push (and hold) the ARM switch, the LED also gets brighter.
Did you face any similar issues while building your samples? Visually everything seems to be ok. Not sure where to start with spotting the faulty part.

colinoflynn, 2022-01-23:
Not sure! I haven't seen that, I've built several of them without issue. It may be a power supply thing it sounds like, you could try breaking the VCC connection between the rpi and power side? Check those voltages as well perhaps?

mehdideveloper, 2022-01-23:
Thanks. Will double check them.
Just putting here for future reference, if someone searches:
Now, after a few resets, pushing the ARM switch turns the ARMING LED on, but as I leave my finger, the LED also turns off (it's ON as long as I press and hold the switch)
Will post my findings and fixes here.

osok, 2022-01-23:
I had a similar problem after making mine.  I put it under a microscope (cheap video one) and saw that there two places where the solder bridged.  A small wire brush actually worked to fix that problem.

mehdideveloper, 2022-01-24:
Ok, so the Raspberry Pico is dead for whatever reason (too much current draw from its 3V3_OUT?!)
I tried on a new bare PCB, but this time I only assembled the micro-controller side. Now when I connect the microUSB, and push the ARM button, the ARMING LED turns on, but the HV LED does not (which makes sense, as the HV side is not populated yet)
Will build the rest and see how it goes.

closed 2022-01-29 by mehdideveloper