newaetech/chipshouter-picoemp #9: Battery holder and pin header compatibility

mehdideveloper, 2022-01-24:
Hey
It seems that either J2 or BH1 are the wrong part number here.
I ordered according to BOM (`S2B-XH-A(LF)(SN)` and `PRT-09925`). They do match, but the polarity is reversed. The VCC on battery holder matches the GND on the pin header/PCB.  The one shown in your Youtube video has the right polarity though. 

Please see the attached pictures:

[image]

[image]

Here's yours. Please note the difference:
[image]

BG7YWL, 2022-01-24:
[image]

colinoflynn, 2022-01-24:
I think yours is defective - see https://www.sparkfun.com/products/9925 which shows the pinout of mine being the same as in the sparkfun photo.

closed 2022-01-24 by colinoflynn

mehdideveloper, 2022-01-24:
Yes. Digikey is selling the wrong product under Sparkfun PRT-09925. Just reported the error to them.

colinoflynn, 2022-01-24:
Might have been an error in that batch, as I bought it in December and it was the "right" way around. If you've got a small screwdriver or anything (nail) it's easy to switch the contacts as in that photo.