newaetech/chipshouter-picoemp #10: Device works but HV LED won't turn on

mehdideveloper, 2022-01-29:
Hi
So I built a second unit and everything seems fine. I measured the voltages, following Colin's Youtube video:
The test header (J3) shows around 228V. When I press the PULSE button, the voltage goes down and again recovers to ~228V.
Also the CHG header measures 2V when the ARMING LED is ON, and 0.77V when it's ON.
And I also tested the device on some STM32 dev boards and a Nintendo Game&Watch, and using the PicoEMP on the STM32, resets the device (actually bricked the Nintendo!)
So apparently it's working as expected.

However, the HV LED never turns on. I press the ARM and ARMING LED turns on, and that's it. It turns off after a minute or so.
Do you have any hints on how I can troubleshoot it?

colinoflynn, 2022-01-29:
Is the voltage on the HV led output OK (should go from 0V to 3.3V roughly for example)? Could be just a regular bad solder etc there if you see the voltage. The easiest way is just change the firmware to turn the HV led on and confirm that works?

A 'todo' was measuring the actual CHARGE pin voltage - basically the current FW uses the digital status of that pin, but you can measure the analog voltage as well to get a better indicator. So it also could be the input voltage isn't being read correctly, the 2V level is around the threshold of high/low IIRC (but that is roughly what it *should* read - so I don't think you've got an error).

Once the firmware supports reading the voltage can just use the analog value instead.

invd, 2022-01-30:
@mehdideveloper : wild guess, is it possible that you've soldered in the HV LED in the reverse orientation?
I've noticed during assembly that the two Kingbright LED color variants have their die on different sides of the package, so that's an easy error if installed "the same way".

mehdideveloper, 2022-01-30:
@invd Well, I double checked. Orientation is correct. I compared it with the other RED LED on the board. Actually in my case, the green and red LEDs are pointed visually the same way.
I measured the voltage across its pins, and there's no voltage, which means (as Colin said) probably the high/low is not being read/set correctly. It's not a big deal though, as the device works as expected.

hexsecs, 2022-06-14:
I'm also having the no HV LED issue. I think the LED itself may be bad.

I wrote a quick test to turn on all of the LEDs on the board. I'm seeing 3.3 V on both GP7 and GP8. 1.9V drop across the status LED (it's on). 3.3 V drop across HV LED. It is off. With board unplugged I tested all of the board LEDs with my multimeter diode tester. Status, Pico LED, and Arm all lit up successfully. No go on HV.

```python
from machine import Pin, Signal

# Status LEDs:
ledHv = Signal(Pin(6, Pin.OUT)) #HV 'on' LED (based on feedback)
ledHv.on()

ledArm = Signal(Pin(27, Pin.OUT)) #Arm 'on' LED
ledArm.on()

ledStatus = Signal(Pin(7, Pin.OUT)) #Simple status LED
ledStatus.on()

ledPi = Signal(Pin(25, Pin.OUT))
ledPi.on()
```