newaetech/chipshouter-picoemp #16: It's possible to shock computer even through USB isolator on target, beware!

hiviah, 2022-02-05:
I was using this isolator - https://www.aliexpress.com/item/4000047053305.html (mentioned in https://github.com/newaetech/chipshouter-picoemp/issues/2#issuecomment-999962799) as intermediate between computer and target (Atmega328), PicoEMP on completely separate power source (since I needed to see USB-serial output to see if glitch was success).

Somehow during the session I managed to shock USB bus through Atmega target connected via the isolator on both computer and monitor USB hubs so they disconnected (power cycling did fix it, so nothing permanent thank god). I am pretty sure I did not short anything, but even in that case the isolator is rated according to docs to 1.5 kV. So this shoudln't have happened, but the quality of the isolator is also question.

There was a powered hub behind the the USB isolator (Atmega<->USB isolator<->powered USB hub<->computer+monitor), but I seriously have no idea which part of the setup could have caused this. Hundreds of glitch attempts were fine until this one.

Even if this was some stupid mistake, there was definitely no direct contact between any parts after the USB isolator.

colinoflynn, 2022-02-05:
The isolators are rated for DC, so indeed these AC pulses may pass through somewhat due to capacitive coupling. The energy in the AC pules should be low enough that (hopefully) the ESD diodes and similar will catch it. The USB interfaces should have good protection there due to how much ESD they are subject to normally (people grabbing the far end of a 1m long USB cable).

But also the USB ports *will* get shut down if they see too many errors at the protocol layer (ask anyone doing USB development at least on Windows... eventually you need to reboot your computer as the ports get turned off). So it's possible for this to happen even if there were no issues with the isolator too, just from glitchy USB traffic.

If you're working with a serial target & want total safety, you might consider using an optoisolator on those serial lines, i.e.:

```
    +----   +5V Isolated -----------+                   
    |                               |
+-----------------+    rx/tx/nRST   |
|    ATMEGA       |     <--- OPTO/ISOLATOR ---> USB-SERIAL CHIP <-----> COMPUTER
+-----------------+
```

hiviah, 2022-02-05:
Thanks!

I've shorted USB bus twice in life by mistake (connecting VBUS +5V to GND accidentally on devboard due to rat's nest wiring), but usually what computer did was disconnect that specific physical connector (bus) and just reconnect fixed it.

In this case all USB physical ports computer+monitor became disconnected until power cycle on all devices which I haven't seen before.

Going to go for opto isolators from now on. Though on datasheet the [ADuM3160](https://www.analog.com/media/en/technical-documentation/data-sheets/adum3160.pdf) on page 6 it says it should handle 560V AC.

On a separate note, would there be a way to repurpose these HDMI optical cables for ETM TRACE as opto isolators? https://twitter.com/jaydcarlson/status/1483162491022712832

colinoflynn, 2022-02-06:
ETM trace is probably going to be tricky, as it seems sensitive to timing even when working normally... tbh I wouldn't worry about isolating the target. I never bother with that and haven't killed anything (you may have a different experience). I *do* add series resistors + ESD/diode clamps though, but basically the actual "injected" voltage isn't likely to be very high (unless you directly short the coil to the target!).

closed 2022-02-06 by hiviah

mrx23dot, 2024-02-27:
would be better to use simple UART with opto isolation or RF instead of usb.