newaetech/chipshouter-picoemp #25: Problem with Build : HV Led wont turn on / Only 7V Voltage

dwelltime, 2022-10-29:
Hello

I have some issues with building this nice device.all parts were used from the bom list except RGT16BM65DT i used instead IKD06N60RATMA1.Gerber files used were rev04_normal.zip also with milled slots for the protection shield.

The solder orientation for the LEDs wasnt so clear to me,so i modified the micropython firmware to run a LED check on startup which works great.

So the actual problem :

The test header (J3) shows only 7V when the ARM button is pressed but the HV LED wont turn on so Pulse doesnt work either.Voltage on (P3) is approx 3V which seems to be correct.Discharge button works.

I looked at the firmware and changed the PWM duty which seems also to work i gained then 12V but its still alot missing hehe

Somewhere i did an error i think but i also rechecked already the solder orientation for the different smd parts.
So can anyone please point me in the right direction and give me some hint?
Should i maybe test the C Firmware? But there is also still some issue with the arming button ive read.

Thanks 

Greetings

tymmothy, 2024-06-27:
Hi - is there any chance that you used a Pi Pico clone? I just debugged a similar situation and tracked it down to the SMPS on the clone not working properly. The board depends on the Pico's SMPS outputting 3.3V; on mine it was putting out the same voltage as the input (which came to about 2.3V after the input protection diode). In my case, it would work as expected when powered from USB, but using a 3V battery pack would lead to the same 7V on the HV rail.

I swapped out the Pico clone for a genuine board and am now seeing arming work as intended and 246V on the HV rail on 3V battery power.

dwelltime, 2024-07-10:
hello!

no i took the official gerber files and made some boards with jlcpcb.then i ordered an picoemp from newatech directly which works.
i still would like to fix the issue :/

tymmothy, 2024-07-10:
Do you mind measuring the voltage on the "3v3(OUT)" pin (P36)? in my case as mentioned I was only seeing about 2.3V (which was too low to properly power the high voltage converter).

Also - have you tried running from USB power?

dwelltime, 2024-07-10:
i think i have the same issue i dont remember correctly i will measure later and report.yes ive also tried to power from usb of course