newaetech/chipshouter-picoemp #34: Discharge button access

gashtaan, 2024-08-05:
I initially thought that when I use exact parts TL3301AF160QJ switch and 1551BTRD shield, it would be somehow possible to press the discharge switch pushing it by pressing the shield above it. The shield fits perfectly to my PCB, but it doesn't work that way - button is too low and/or shield too sturdy. How was it intended?

colinoflynn, 2024-08-06:
This is intended - when you remove the shield, you need to manually push the discharge button yourself. Basically it's a nicer way of discharging, even if not fully foolproof. You can drill a small hole in the shield & then use a small dowel or similar to discharge it is basically the best "fully safe" way.

The ceramic caps don't hold their charge too long compared to e.g., a Xenon flash capacitor, so ultimately most of the time it does drain off unless you are removing the shield shortly after having it on (which hopefully means you remember to discharge it first!).

Using a NC button that has a longer pluged so it automatically discharges was a possability, but it became an issue that (1) NC isn't readibly available in pushbuttons, and (2) it can be helpful for debug running with the shield off (or doing voltage vs. PWM curves).

closed 2024-08-06 by colinoflynn

gashtaan, 2024-08-06:
So I've overthinked it a bit :) I just thought that button will be possible to press through the shield (be deforming it a little).