newaetech/chipshouter-picoemp #37: Ticking sound

gashtaan, 2024-09-07:
Anyone experiencing quite loud ticking sound coming from somewhere around C3 capacitor, Q2 transistor when firing the pulse? It's difficult to pin-point the exact location, I rather not to keep my ear too close to this high voltage circuit :)

I suspect some pieco-electric effect in capacitor. I chosen GRM55DR7LV474KW01K for this part.

colinoflynn, 2024-09-09:
You are correct - the capacitor will do that! It happens with the discharge due to the fast discharge current / voltage change. In fact the larger ChipSHOUTER capacitors have little metal "tabs" to help absorb some of the movement as you can otherwise cause solder joint damage from the movement of the caps over many discharge cycles (the ChipSHOUTER caps are larger & have a higher current change).

gashtaan, 2024-09-09:
@colinoflynn Will use of capacitor with flexible termination less noisy? Or there is nothing to do about it?

colinoflynn, 2024-09-10:
I'm not sure if less noisy - the noise isn't something I guess I was worried about before, normally I think it's a good thing as you know it's working!

gashtaan, 2024-09-10:
> normally I think it's a good thing as you know it's working!

That's true, initially I thought the same. Although at first I thought it's caused by movement of loose wires in the coil which I can easily solve. The problem is that I want to use it in some automation over extended period of the time, but because of that ticking I can't keep it running overnight or during baby's naps :)

Anyway, at least I know the noise is normal and expected. Thanks for the info.

closed 2024-09-10 by gashtaan

colinoflynn, 2024-09-10:
Some movement of the loose wires in the coil also causes sound, this is more obvious in the larger ChipSHOUTER. If noise is a problem I'd try dabbing a bunch of silicon sealant around the capacitor (ideally something that is designed for electronics, *not* stuff that smells like vinegar as that smell is acetic acid & will cause rapid corrosion). You could try some hot glue or something else too. Even a small piece of foam maybe... the cap won't have too much heat, you may want to leave a little room for the coil to have heat dissipation, but even that shouldn't be too terrible.

gashtaan, 2024-09-27:
@colinoflynn I just swapped the C3 capacitor to KRM55TR72J474MH01K (the one you've recommended in BOM parts). It still makes the ticking sound, but now it's hardly noticeable. The difference is enormous. I shouldn't have experimented with it :)